vaslabs/pi-web-agent #117: Tool to construct packages for the various supported distros

vaslabs, 2015-01-05:


agalazis, 2015-03-21:
Is there any similar tool ? One idea is to have different folders with the installation slit into parts eg:
install
|_pidori/websocketdBro
|_rspbian/websocketdBro

then in the script
DSITRO = $(determinedistro)
and when we want to install we will use:
./install/$DSITRO/websocketdBro        #or even define functions for each component

vaslabs, 2015-03-21:
We want rpm and deb packages that will have all the information including dependencies

agalazis, 2015-05-21:
I suggest closing this issue the purpose  of the other issues is to make the scripts for building   the suggested packages(not just building manually once)

vaslabs, 2015-05-21:
Didn't quite get that...

vaslabs, 2015-05-21:
This issue is an epic, it considers all packaging related issues. Of this was jira, every related issue would be linked to this epic

closed 2015-07-09 by vaslabs