vaslabs/pi-web-agent #131: Some users complaint about the rabbit mq. Make installation of mplayer dependencies optional.

vaslabs, 2015-04-04:
This is quite urgent

agalazis, 2015-04-04:
Ok

agalazis, 2015-04-04:
My sincerest apologies about rabbitMQ,what was the problem the philosophy behind making it a system-wide dependency is than any application can use it's own channel. If we have a system wide dependency  we won't end up having different flavors of message ques in every application. Ofcourse am not sure what is the exact problem they refer to... but a quick solution is to make it optional

agalazis, 2015-04-04:
I believe that there is no problem to remove it as any application that will use it can have it as a depndency as well

vaslabs, 2015-05-21:
This issue will be closed because the users that complaint didn't give enough information and no other complaints have been filed.

closed 2015-05-21 by vaslabs