vaslabs/pi-web-agent #136: IndieCity dead -- how to install?

stuzog, 2015-10-22:
With IndieCity now dead, how can pi-web-agent be installed on a Pi?

vaslabs, 2015-10-23:
https://github.com/vaslabs/pi-web-agent/releases

stuzog, 2015-10-23:
Thanks, Vasilis!

stuzog, 2015-10-23:
> Vasilis Nicolaou <mailto:notifications@github.com>
> October 23, 2015 at 7:27 AM
>
> Closed #136 <https://github.com/vaslabs/pi-web-agent/issues/136>.

Unfortunately, it stil doesn't work. After installation, I get:

$ sudo /etc/init.d/pi-web-agent start
Starting pi web agent [ FAILED ]

Following the instructions, the directory structure set up in / is:

Is this correct?

Stuart

vaslabs, 2015-10-23:
can you do an ls on your current directory that you are running ./install.sh and paste what you get here?

vaslabs, 2015-10-23:
Also could you send me the error logs that are in /var/log/pi-web-agent

stuzog, 2015-10-23:
> October 23, 2015 at 8:41 AM

pi@raspberrypi /pi-web-agent0.4.2/0.4.2 $ ls
AUTHORS   etc         LICENSE  run.sh        setup_dev.sh  uninstall.sh
build.sh  install.sh  README   setup_common  setup.sh      usr

Stuart

vaslabs, 2015-10-23:
that looks alright, send me the error log in /var/log/pi-web-agent to see where the apache failed

stuzog, 2015-10-23:
Hi Vasilis,

There is no error log, just service.log:

pi@raspberrypi /var/log/pi-web-agent $ ls
service.log
pi@raspberrypi /var/log/pi-web-agent $ cat service.log
apache2: Syntax error on line 106 of 
/etc/pi-web-agent/conf/apache2-appliance.conf: Cannot load 
modules/mod_authn_alias.so into server: 
/etc/pi-web-agent/modules/mod_authn_alias.so: cannot open shared object 
file: No such file or directory

Stuart

vaslabs, 2015-10-23:
oh dear, it looks like apache got updated.
We haven't worked on pi-web-agent for a long time...
I guess the solution would be to open /etc/pi-web-agent/conf/apache2-appliance.conf and comment out the modules that are not used but I think there will be problems with some keywords as well.

vaslabs, 2015-10-23:
or downgrade to apache 2.2

stuzog, 2015-10-23:
Unfortunately, that's beyond my ability right now, I'm new to Linux.

S.

vaslabs, 2015-10-23:
ah, sorry mate. I hope I'll find some time to find a fix but I'm struggling with other projects.

stuzog, 2015-10-23:
OK, I understand. It's not mission-critical for me.
Thanks for your help and especially for your quick response,
All the best,
Stuart

vaslabs, 2015-10-23:
Out of interest, I know the apache config is not working with ubuntu. What distro have you installed?

stuzog, 2015-10-23:
Raspbian Jesse (Debian 8.0)

vaslabs, 2015-10-23:
Thanks. I'll install it when I have time and make a fix

agalazis, 2015-11-15:
well I was thinking to make a release on 1st feb 
We can put all our efforts to improve install process first

vaslabs, 2015-11-16:
Yes, if you can write some guidelines how we use what's in the current master branch, then we'll all be at the same page

stuzog, 2015-11-16:
Hi Vasilis,

I'm not quite sure what you mean, but I installed in the pi user 
directory as per the link in your email. It started with:

pi@raspi ~/pi-web-agent0.4.2/0.4.2 $ sudo ./install.sh
Installing pi web agent [ OK ]
initialising submodules ...
fatal: Not a git repository (or any of the parent directories): .git
[ OK ]
Updating packages
....[etc]

It seemed to install everything:
All Done.

NOTE: To compile programs with wiringPi, you need to add:
-lwiringPi
to your compile line(s) To use the Gertboard, MaxDetect, etc.
code (the devLib), you need to also add:
-lwiringPiDev
to your compile line(s).

DONE

But ended with:

Registering pi-web-agent in sudoers
Starting pi-web-agent
Starting pi web agent [ FAILED ]
/home/pi/pi-web-agent0.4.2/0.4.2
pi@raspi

Also:

pi@raspi ~ $ sudo /etc/init.d/pi-web-agent start
Starting pi web agent [ FAILED ]

Stuart

vaslabs, 2015-11-16:
Hi,
Sorry, my response was for @agalazis . you don't need to do anything else, once we fix it this ticket will be closed. But it will take a while since everyone seems busy.

vaslabs, 2016-02-25:
Hello @stuzog .
I'm happy to let you know that I've just released pi-web-agent for raspbian jessie
https://github.com/vaslabs/pi-web-agent/releases/tag/0.4-rc-2_jessie
Let me know for any issues.

closed 2016-02-25 by vaslabs