vaslabs/pi-web-agent #137: redefinition of ‘struct sysinfo’ in live_info.c

agalazis, 2016-02-07:
while i was trying to build a new version I had this error:
```
live_info.c:1:26: fatal error: proc/sysinfo.h: No such file or directory
 #include "proc/sysinfo.h"
```
so I decided to install `libprocps3-dev`

but afterwards I was getting:
```
/home/pi/pi-web-agent/src/pi-web-agent-0.4.3
live_info.c:21:8: error: redefinition of ‘struct sysinfo’
 struct sysinfo {
        ^
In file included from /usr/include/linux/kernel.h:4:0,
                 from /usr/include/linux/netlink.h:4,
                 from /usr/include/linux/if_link.h:5,
                 from live_info.c:16:
/usr/include/linux/sysinfo.h:7:8: note: originally defined here
 struct sysinfo {
```
@vaslabs  Any Ideas on the best approach to fix this? is it  safe to remove the struct from live_sysinfo.c?

vaslabs, 2016-02-07:
In some older versions of Debian it wasn't compiling so it needed it. You can remove it yes.

vaslabs, 2016-02-25:
I've removed this in the jessie work

closed 2016-02-25 by vaslabs