vaslabs/pi-web-agent #141: Responsive UI

vaslabs, 2016-02-25:
I'm initially assigning it to you. We should completely redesign the UI.

agalazis, 2016-02-26:
True that I thought of it before although even at work I am using backbone at a widget level. I think it would be nice to move on with webcomponents on this project but we 'll need to create something clever that every developer can work with independently of their background.The main issue is re-usability of ui components + organizing/building/ optimizing custom ui code . With the current problems responsive ui becomes a secondary issue, but we can have it in mind while we rebuild ui development process

vaslabs, 2016-02-26:
no, first lets redesign the UI to be responsive. Remove bootstrap and put foundation3 instead, I like more the class names. We can also use material design. Forget about plugins for now, although we should be able to separate js from css.

vaslabs, 2016-02-26:
maybe webcomponents is a good idea

vaslabs, 2016-04-03:
branch : https://github.com/vaslabs/pi-web-agent/tree/materializeDesign

closed 2016-04-10 by vaslabs