vaslabs/pi-web-agent #153: my server stop working

platini76, 2017-11-21:
I have again problem with server working... without any changes, only a package update.

vaslabs, 2017-11-21:
Any errors? Which package?

platini76, 2017-11-21:
> Risposta: vaslabs/pi-web-agent <reply@reply.github.com>
> Data: martedì 21 novembre 2017 11:51

Yes… my pi manager don’t work.. no page

agalazis, 2017-11-21:
hello which version of rpi do you use so we can have a look (@vaslabs should I add an issue template?)

vaslabs, 2017-11-21:
@agalazis yes please, we really need more information on issues otherwise we spend time clearing things out and I don't like deleting issues just because they lack information.

platini76, 2017-11-21:
Here screenshot

Data: martedì 21 novembre 2017 18:02

platini76, 2017-11-29:
HELLO?

Are you there friends?

Thanks for help me.

davide

Data: martedì 21 novembre 2017 17:25

agalazis, 2017-11-29:
@platini76  I don'ts see any screenshot mate  just give us rapbery board version(I have both v2 and v3) + raspbian version. @vaslabs  I will do this afternoon

platini76, 2017-11-29:
I have no access! I cant’ make a screen!

Data: mercoledì 29 novembre 2017 07:53
Cc: platini76 <info@dvd24h.it>, Mention <mention@noreply.github.com>

platini76, 2017-11-29:
Maybe the last update corrupt installation:

 

Esamina Porte avviato…

 

Port Scanning host: 192.168.0.10

 

       Open TCP Port:       22                  ssh

       Open TCP Port:       80                  http

       Open TCP Port:       83                  mit-ml-dev

       Open TCP Port:       4369                epmd

       Open TCP Port:       5672                amqp

       Open TCP Port:       5901

       Open TCP Port:       6001

       Open TCP Port:       12122

       Open TCP Port:       12133

       Open TCP Port:       12200

       Open TCP Port:       12201

       Open TCP Port:       12202

       Open TCP Port:       12203

       Open TCP Port:       12204

       Open TCP Port:       25672

Esamina Porte completato…

 

 

 

Da: "info@dvd24h.it" <info@dvd24h.it>
Data: mercoledì 29 novembre 2017 08:17
A: vaslabs/pi-web-agent <reply@reply.github.com>
Oggetto: Re: [vaslabs/pi-web-agent] my server stop working (#153)

 

I have no access! I cant’ make a screen!

platini76, 2017-11-29:
Data: mercoledì 29 novembre 2017 01:35
A: vaslabs/pi-web-agent <reply@reply.github.com>, vaslabs/pi-web-agent <pi-web-agent@noreply.github.com>
Cc: Author <author@noreply.github.com>

platini76, 2017-11-29:
Problem partially solved… the service don’t start automatically, is there a command to make it start?

Data: mercoledì 29 novembre 2017 08:54

vaslabs, 2017-11-29:
Can you send me the logs? They should be in /var/log/pi-web-agent/ssl_* . They might have changed the way services are registered in raspbian or it was disabled because of incompatibility issues, we don't actively follow the OS changes anymore, we wait for issues to come so we need to investigate into yours and figure it out.
I'll also need your board version and the information from running the command `uname -a`

platini76, 2017-12-03:
SOLVED, simply I reinstall the app. 

thanks

Data: mercoledì 29 novembre 2017 11:37

agalazis, 2017-12-03:
Interesting not sure if we should close it though, the fact that you needed to reinstall it is an issue itself did you upgrade from an older version or did you perform a clean installl in first place?

platini76, 2017-12-04:
In past I upgrade… but 1 year ago..

Data: domenica 3 dicembre 2017 21:21

platini76, 2017-12-19:
Some problems in reinstall 

readall.c: In function 'abReadallJSON':

readall.c:431:11: warning: variable 'type' set but not used [-Wunused-but-set-variable]

     char *type ;

           ^~~~

[Link]

[Install]

 

All Done.

 

NOTE: To compile programs with wiringPi, you need to add:

    -lwiringPi

  to your compile line(s) To use the Gertboard, MaxDetect, etc.

  code (the devLib), you need to also add:

    -lwiringPiDev

  to your compile line(s).

 

DONE

Registering pi-web-agent in sudoers

setup_common: riga 144: a2enmod: comando non trovato

Starting pi-web-agent

 not installed

 Restoring Camera Snapshots

Recovering your snapshots

 

 

Can you help me?

thanks

 

Da: "info@dvd24h.it" <info@dvd24h.it>
Data: lunedì 4 dicembre 2017 01:52
A: vaslabs/pi-web-agent <reply@reply.github.com>
Oggetto: Re: [vaslabs/pi-web-agent] my server stop working (#153)

 

In past I upgrade… but 1 year ago..

platini76, 2017-12-19:
Solved, but now there is another error:

Bad Request

Your browser sent a request that this server could not understand.
Reason: You're speaking plain HTTP to an SSL-enabled server port.
Instead use the HTTPS scheme to access this URL, please.

Apache/2.4.25 (Debian) Server at raspberry.fritz.box Port 443

 

 

Da: "info@dvd24h.it" <info@dvd24h.it>
Data: martedì 19 dicembre 2017 13:36
A: vaslabs/pi-web-agent <reply@reply.github.com>
Oggetto: Re: [vaslabs/pi-web-agent] my server stop working (#153)

 

Some problems in reinstall 

readall.c: In function 'abReadallJSON':

readall.c:431:11: warning: variable 'type' set but not used [-Wunused-but-set-variable]

     char *type ;

           ^~~~

[Link]

[Install]

 

All Done.

 

NOTE: To compile programs with wiringPi, you need to add:

    -lwiringPi

  to your compile line(s) To use the Gertboard, MaxDetect, etc.

  code (the devLib), you need to also add:

    -lwiringPiDev

  to your compile line(s).

 

DONE

Registering pi-web-agent in sudoers

setup_common: riga 144: a2enmod: comando non trovato

Starting pi-web-agent

 not installed

 Restoring Camera Snapshots

Recovering your snapshots

 

 

Can you help me?

thanks

 

Da: "info@dvd24h.it" <info@dvd24h.it>
Data: lunedì 4 dicembre 2017 01:52
A: vaslabs/pi-web-agent <reply@reply.github.com>
Oggetto: Re: [vaslabs/pi-web-agent] my server stop working (#153)

 

In past I upgrade… but 1 year ago..

vaslabs, 2017-12-19:
What is the url you put in your browser?

platini76, 2017-12-19:
Ip address:8001

Data: martedì 19 dicembre 2017 14:23

vaslabs, 2017-12-22:
That doesn't look right, it should be https://IP:8003

closed 2018-02-14 by vaslabs